tillitis/tillitis-key1 #2: Add support for USS

secworks, 2022-09-20:
The current version of the FW and host side does not support including a USS. The protocol, the FW and the host SW needs to be updated to handle a USS.

quite, 2022-09-20:
Exactly that type is `USS`? Looks like a string in tpt. Any max length then?

quite, 2022-09-22:
Draft implementation of this in `uss` branch in this and the apps repo. See the above references.

Discussion points:

- The (user input string for the) secret USS should probably not be passed as a cmdline arg. So ask user to to type it then? Can also read it from file passed on cmdline, with classic "-" meaning stdin.

- But how to make this usable in mkdf-ssh-agent?
 
- Do users need to understand that the string they input (by some means) is not used directly, but is hashed to become the USS (32 bytes)?

secworks, 2022-09-22:
IMHO one should look at the USS (and treat is as such) as a secret. But it can potentially be used as a context parameter similar in a KDF. Compare to how the user supplied entropy works in the tpt tool. This means that as comments to point one, the USS could be passed either on a cmdline or interactively (what TPT supports)

Regarding what the user understands, I see that more of a documentation, UX issue. If the documentation states that what you input as USS will be hashed before sent to the device, the user could be expected to understand. One can of course also display the generated hash, to allow the user to see (and verify) the hash being sent and verify that it is correct.

mc-mullvad, 2022-09-27:
Firmware code for USS handling looks good to me.

I think the host part should be rebased on top of the I/O refactoring which probably should be merged first.

Looked through any way of using the ssh-agent to send a USS and found nothing. Had some idea if we had some context data about the use of the key for every signing but there's no such thing in the interface. I suggest we just use a zeroed USS when using the agent for now.

quite, 2022-09-28:
I've added interactive typing of "USS phrase" in runapp (if `--uss` is passed). It all works and gives different pubkeys in the end.

About the ssh-agent, yes that was my concern -- how to make USS useful for it. There is no context to receive, so user would have to run separate mkdf-ssh-agent, each with different USS. But! We can't have multiple connections to the serial port... so no go. Yes, we should just not send USS from the ssh-agent -- the result is zeroed USS in firmware.

secworks, 2022-09-28:
Having the --uss as a parameter to the runapp seems to me to be the correct thing to do. If an application wants other things from the user (challenge, Time of Day, password), it is something it needs to solve itself. But having a --uss for runapp provides addtional security, and allow for context aware loading. So I like this.

Using --uss "ssh for host XYZ" when loading the ssh app looks like a nice way for me to use the same app for different hosts. The security is still provided by the UDS and the measurement.

quite, 2022-09-28:
Hm, yes. One could let the ssh-agent take uss in that manner. And then expect the user to stop and restart the ssh-agent when they want to change "identity".

mchack-work, 2022-09-29:
Merged.

closed 2022-09-29 by mchack-work