tillitis/tillitis-key1 #4: tillitis-key1/hw/boards/mta1-usb-v1/ USB-C connector might not be very durable?

GvY85, 2022-09-20:
Hi,

First of all, thanks for the great work. 

If I may, I would like to critique the USB-C connector choice a bit. It seems you are limited (but then you did your own housing design) in PCBA space but it feels like the SMT USB-C connector uses is not very durable and would come off the board pretty quickly after some insertion cycles.
Any chance of changing it to a part that offers some THT mechanical support? This can even be a pin-in-paste type so that you do not need an additional THT solder step but can reflow the THT connector part.

Also, I noticed a few instances in the layout where the min gap is around 100um that could easily be extented to 125um minimum track-gap to improvbe manufacturability a bit.

cibomahto, 2022-09-20:
Thanks- do you have any recommendations for a connector?

The plastic housing is designed to capture the USB connector, so that the majority of the insertion/removal forces are absorbed by the case/connector interface, instead of the connector/pcb interface.

GvY85, 2022-09-20:
Hi,

Thanks for the quick reply. 
I understand the reasoning behind it but in my experience in the long term that is not always the most robust solution.

Some connectors can be found here, most are reasonably priced (<€1,-) and feature 4 THT solder tabs that, with creative solder paste design, can be soldered in a SMT reflow process

[Digikey](https://www.digikey.nl/en/products/filter/usb-dvi-hdmi-connectors/usb-dvi-hdmi-connectors/312?s=N4IgjCBcoEwBxVAYygMwIYBsDOBTANCAPZQDaIALGHAMwCcArCALqEAOALlCAMocBOASwB2AcxABfQjDqIQKSBhwFiZEPRkA2AAwt2XSLwEjxU9XRhyFSvIRKRyVOBU0JWITtz5CxkwmDo4WWh5NCxbVQcQXWYzAFpLEIUBAFcVe3ImWLNNOUEAE244sG0IfW4QQg4ATzZcbnRsFAkJIA)

GvY85, 2022-09-20:
Hmm, I see I linked in receptacles and not plugs.

Plug doe not seem to have such robust options it seem....:

[Digikey](https://www.digikey.nl/en/products/filter/usb-dvi-hdmi-connectors/usb-dvi-hdmi-connectors/312?s=N4IgjCBcoEwBxVAYygMwIYBsDOBTANCAPZQDaIALGHAMwCcArCALqEAOALlCAMocBOASwB2AcxABfQjDqIQKSBhwFiZEDTgB2OAAYdLdl0i8BI8VJABaGHIUCAripKRyTZhItNoINlDDtfSBgGCwA2OUEAE25LMB0IQ24QQg4ATzZcbnRsFA8gA)

closed 2022-11-21 by cibomahto