tillitis/tillitis-key1 #10: Flash the key from the container (a rootless podman container)

faern, 2022-09-25:
I have been trying to figure out how to flash the device from within the [build container](https://github.com/tillitis/tillitis-key1/blob/main/contrib/Dockerfile) without needing root. This issue serves both as documentation for others coming here, and a request to update the documentation around it :)

1. Add udev rules to allow the `dialout` group to access the programmer board:
    ```
    $ cat /etc/udev/rules.d/55-tillitis.rules
    SUBSYSTEM=="usb", ATTR{idVendor}=="cafe", ATTR{idProduct}=="4004", MODE="0666", GROUP="dialout"
    KERNEL=="hidraw*", ATTRS{idVendor}=="cafe", ATTRS{idProduct}=="4004", MODE="0666", GROUP="dialout"
    
    $ sudo udevadm control --reload-rules && sudo udevadm trigger
    ```
    Make sure to unplug and plug your device again after this.

1. Build the container:
    ```
    cd contrib/; podman build -t tillitis-dev .
    ```

1. Build the firmware and flash the device, all from within the container:
    ```
    podman run --rm --security-opt label=disable \
        --device /dev/bus/usb/$(lsusb | grep -m 1 cafe:4004 | awk '{ printf "%s/%s", $2, substr($4,1,3) }') \
        -v .:/build:Z -w /build/hw/application_fpga \
        -it tillitis-dev make prog_flash
    ```
    If you do not have SELinux you can probably drop the `--security-opt label=disable` and `:Z` part. I have reported an issue to `podman` on this. https://github.com/containers/podman/issues/15930.

faern, 2022-09-26:
I got some very helpful tips in https://github.com/containers/podman/issues/15930 that allowed me to flash without disabling SELinux :partying_face:

mchack-work, 2023-09-11:
Thanks for the input. Now documented in Dev Handbook at https://dev.tillitis.se/ - only a year later!

closed 2023-09-11 by mchack-work