tillitis/tillitis-key1 #16: verilator broken by 90a57c4

quite, 2022-10-03:
90a57c4 on the pll branch broke compilation `make verilator`. But perhaps it had issues before that..

```
%Error: /home/quite/t/tillitis-key1/hw/application_fpga/tb/application_fpga_vsim.v:172:3: Cannot find file containing module: 'reset_gen'
  172 |   reset_gen #(.RESET_CYCLES(200))
      |   ^~~~~~~~~
%Error: /home/quite/t/tillitis-key1/hw/application_fpga/tb/application_fpga_vsim.v:172:3: This may be because there's no search path specified with -I<dir>.
  172 |   reset_gen #(.RESET_CYCLES(200))
      |   ^~~~~~~~~
        ... Looked in:
             reset_gen
             reset_gen.v
             reset_gen.sv
             verilated/reset_gen
             verilated/reset_gen.v
             verilated/reset_gen.sv
%Error: Exiting due to 2 error(s)
make: *** [Makefile:155: verilator] Error 1
```

secworks, 2022-10-03:
Darn. The reset_gen became the clk_reset_gen, since the top level became a bit more complex when adding PLL and BUFG. I need to add a verilator specific reset_gen.

secworks, 2022-10-03:
Fixed the issue by adding a separate reset generator in the tb directory. I also changed names on Verilog source files groups in the Makefile to make it more distinct that there are some source files specific to building the FPGA, some source files that are specific for the Verilator simulation, and some source files that are common to both targets.

See https://github.com/tillitis/tillitis-key1/commit/b2ca3f2ea020df86397e3ee2c8af94ea45abf163?diff=unified for the commit details.

closed 2022-10-03 by secworks