tillitis/tillitis-key1 #18: Adjust Post build analysis parameter

airskywater, 2022-10-07:
icetime -c 12 need to be changed to -c 18

secworks, 2022-10-12:
Ouch! Thanks!

secworks, 2022-10-12:
icetime really don't like combinational loops. Need to check if one can ignore those paths.

Report for critical path:
-------------------------

        loop-start at net_93660
        t26053 (LocalMux) I -> O: 1.099 ns
        inmux_24_27_97944_97979 (InMux) I -> O: 0.662 ns
        lc40_24_27_5 (LogicCell40) in1 -> lcout: 1.232 ns
1000002.993 ns net_93660 (trng_inst.f[11])
        t26052 (LocalMux) I -> O: 1.099 ns
        inmux_24_27_97936_97949 (InMux) I -> O: 0.662 ns
        lc40_24_27_0 (LogicCell40) in1 -> lcout: 1.232 ns
1000005.987 ns net_93655 (trng_inst.rosc_ctrl_logic.xor_f_SB_LUT4_O_I0[2])
        odrv_24_27_93655_93688 (Odrv4) I -> O: 0.649 ns
        t26058 (LocalMux) I -> O: 1.099 ns
        inmux_24_29_98193_98260 (InMux) I -> O: 0.662 ns
1000008.397 ns net_98260 (trng_inst.rosc_ctrl_logic.xor_f_SB_LUT4_O_I0[2])
        lc40_24_29_7 (LogicCell40) in0 [setup]: 1.060 ns
1000009.457 ns net_93908 (trng_inst.sample1_new[1])

Resolvable net names on path:
1000002.993 ns ..1000004.755 ns trng_inst.f[11]
1000005.987 ns ..1000008.397 ns trng_inst.rosc_ctrl_logic.xor_f_SB_LUT4_O_I0[2]
                  lcout -> trng_inst.sample1_new[1]

Total number of logic levels: 3
Total path delay: 1000009.46 ns (0.00 MHz)

// Checking 55.56 ns (18.00 MHz) clock constraint: FAILED.

commit 192ce47f (2022-10-17): Fix #18 with incorrect clock frequency in analysis
closed 2022-10-12 by secworks