tillitis/tillitis-key1 #23: Move to the zmmul subset of the RISC-V M extensions

quite, 2022-10-18:
This will allow us to bring in the PicoRV32 with ENABLE_DIV=0, and use the freed resources in the FPGA for other things.

This will require clang-15. Is this sensible to do until Ubuntu 22.04 LTS gets clang-15 -- will it get it?

mchack-work, 2022-10-18:
For Debian packages (also usable in Ubuntu) of clang one can use https://apt.llvm.org/ but of course requires that you trust another package repo. Might want to wait until at least Ubuntu gets it.

quite, 2022-11-15:
May be helpful: https://github.com/riscv/riscv-isa-manual/issues/869

quite, 2022-11-23:
The correct -march seems to be `-march=rv32iczmmul`. I have tried adding a DIV in our inline asm. With `imc` it compiles, with `iczmmul` clang complains that it needs 'M'.

Do we need a test with C code that really produces a DIV on `imc`, but not on `iczmmul`?

mchack-work, 2022-11-29:
Zmmul changes to firmware and apps merged.

closed 2022-11-29 by mchack-work