tillitis/tillitis-key1 #28: Change name to TK1 everywhere

mchack-work, 2022-10-20:
There are a lot of MTA1 and MTA1-MKDF everywhere in the code and filenames.

@quite started branch tk1 in:
- qemu
- tillitis-key1
- tillitis-key1-apps

mchack-work, 2022-10-20:
The Go modules are still called "mkdf" and "mkdfsign" in tillitis-key1-apps. Do we change them even if it means we break some external software? We don't want to tag with a v1 for Go tagging reasons.

quite, 2022-10-20:
@mchack-work I think we should change them, and getting the new name somewhat right is more important that worrying about breaking (any) existing software.

We're building ourselves into the name `tk1` now in general... but maybe that's ok, and the package could be `tk1` and `tk1sign`? Or just `tk` prefix? I'm not sure of the benefit; the repo is tillitis-key-*1*...

quite, 2022-10-21:
I added a commit that renames the go pkgs to tk1 and tk1sign. Also changes the app internal names from `mkdfFOOB` to `tk1 FOOB`.

quite, 2022-10-21:
I think that leaves only hw/boards/ in tillitis-key1.

mchack-work, 2022-10-21:
Using `tk1` and `tk1sign` for the Go modules works for me. I agree we should change them. We should also probably warn people at least in the chat.

quite, 2022-10-25:
PRs: https://github.com/tillitis/tillitis-key1/pull/35  https://github.com/tillitis/tillitis-key1-apps/pull/14

For qemu we would continue just using the tk1 branch: https://github.com/tillitis/qemu/tree/tk1

secworks, 2022-10-26:
While we are discussing naming things - should it be "tk1", or "tk"? ;-)

The repos right now is for the current key, TK1. But will we create new repos for the next generation, or just continue, creating releases? Having a version in the names is ok, good? Clang, LLVM seems to think so. OpenSSL for example does not change the names of things. (The OpenSSL versioning scheme is another matter.)

"There are 2 hard problems in computer science: cache invalidation, naming things, and off-by-1 errors."

quite, 2022-10-26:
We did it!

closed 2022-10-26 by quite