tillitis/tillitis-key1 #30: Use TK1 for Git signatures

mchack-work, 2022-10-21:
We might be able to use TK1 for our own Git signatures.

How do we use our own ssh-agent with Git?

We might want to include an `allowed_signers` file in the repos so that Git commits can be verified against it.

quite, 2022-10-24:
I got it working. Didn't need to modify SSH_AUTH_SOCK in the shell, but created a wrapper program that git is told to use (in two instances). It's a spaghetti a la shell, check it out:

https://github.com/tillitis/tillitis-key1-apps/blob/ssh-signing/setup-ssh-signing-and-start-agent

quite, 2022-10-25:
We might want to have basic #33 process for provisioning in place before starting to do this? So we have the same way of setting up our live signing devices.