tillitis/tillitis-key1 #38: Stack size?

mchack-work, 2022-10-26:
The size of the stack controls how large apps we can load into memory.                                                                      
The current stack is huge at 64 k and we have only 64 k to load the                                                                         
app into.                                                                                                                                   
                                                                                                                                            
I have done experiments with as low as 8 k stack and it still works,                                                                        
including the signerapp. If we go this path we can load apps that are                                                                       
120 k.                                                                                                                                      
                                                                                                                                            
A memory map like this	is a bit	complicated, though, since a smaller                                                                        
app then will have free	memory	both above and below itself. If an app                                                                       
wants to move itself in memory to get more unfragmented free memory to                                                                      
play with we provide `APP_ADDR` and `APP_SIZE` which should be	enough                                                                       
for the app to move itself, right?                                                                                                          
                                                                                                                                            
I suggest we just decide on a stack size, say 8 or 16 k, and run with                                                                       
that. Perhaps also write an example app that moves itself around in                                                                         
memory.                                                                                                                                     
                                                                                                                                            
@quite says long ago in another organization:                                                                                                
                                                                                                                                            
When the signer app has just returned the signature to its client,                                                                          
some 6380 bytes of stack has been written to -- from 0x8000ffff and                                                                         
down (as set in crt0.S). Tested this by dumping the stack mem at this                                                                       
point, and also inserting some deadbeefs in mem to see exactly what                                                                         
got overwritten.                                                                                                                            
                                                                                                                                            
Also noting that when app main is entered, 691 bytes (i think) of                                                                           
stack has already been used. Because the first stack var in main named                                                                      
stack is at 0x8000fd48 . Not sure who eats that, the C runtime?

quite, 2022-10-31:
@mchack-work 
I've worked on this today. Draft PRs here https://github.com/tillitis/tillitis-key1/pull/40 and https://github.com/tillitis/tillitis-key1-apps/pull/18

I'm trying 8 kb stack, and it mostly works. The issue I see is that the setup of new stack doesn't seem to bite. https://github.com/tillitis/tillitis-key1-apps/pull/18/files#diff-3857988385a284fb03dc747adb79566dac6d15e071f51256f9a7e702e4208d62R84

`li sp, NEW_STACK` might not be enough? The second time the program puts `&stack`, the addr is the same as the first. Some proper relocation needed? But how?

Might there then be other issues due to not doing proper relocation? I'm not so familar with riscv addressing yet, could there be address references that needs to be relocated? But the jump to `forever_ledflash()` works at least...

mchack-work, 2022-11-01:
We probably can't do any relocation as inline assembler. If so, we're messing with the C runtime. 

In our relocation example we want to run the relocation *before* starting running C code. This probably means linking with another libcrt0 if people want support for relocation.

quite, 2022-11-02:
Ah yes, I tried my hand at accomplishing relocation by having a different app.lds and crt0.S. Relocating the init crt0 at 0x40002000 and the app C main at 0x40010000. Setting the stack below that in crt0.S. It almost worked, except the `call main` ended up executing a jump to 0x40012000 -- too high. But it feels like this could be possible.

I think I'll leave the example app outside scope of this issue. And also settle on 16 kb stack, 8 seems a bit too small

quite, 2022-11-03:
Closed with these:
https://github.com/tillitis/tillitis-key1-apps/pull/18
https://github.com/tillitis/tillitis-key1/pull/40

closed 2022-11-03 by quite