tillitis/tillitis-key1 #39: Increase size of RX-FIFO

secworks, 2022-10-28:
The RX FIFO in the UART currently has a capacity of 256 bytes. The EBR allocated for the RX-FIFO however is able to store 512 bytes. Increasing the FIFO to 512 bytes would allow at least three complete frames to be stored, increasing the number of frames in flight and better throughput. Getting performance increase for "free" just by using the whole EBR is nice.

The following branch contain a change to the FIFO to support 
https://github.com/tillitis/tillitis-key1/tree/bigger_rx_fifo

quite, 2022-10-29:
I run test test-loop at our current 62500, it did as good as before, >120k iterations .

secworks, 2022-11-02:
So we could merge this.

secworks, 2022-11-02:
Merged in https://github.com/tillitis/tillitis-key1/commit/517fafff57d936e85dae5eed54b7a17eb5fd3abc

closed 2022-11-02 by secworks