tillitis/tillitis-key1 #49: Lock down access to UDI to only FW

secworks, 2022-12-07:
Only the FW should have access to the UDI.
We should add a check of the switch_app_reg when handling UDI read operations.

quite, 2022-12-07:
Oh, is having app access it some security issue?  In tkey-ssh-agent it is useful to display as the ssh key comment in `ssh-add -L`, so use can identify which TKey is plugged in that way.

quite, 2022-12-07:
Firmware protocol should still allow fetching of UDI, right?

secworks, 2022-12-08:
Yes, FW will have access to UDI. It is useful to display UDI to the user when the device has been connected. It is also useful for Tillitis to be able to extract UDI to identify devices during production, handling of warranty, repair issues.

secworks, 2022-12-08:
The security issue with apps having access to the UDI allows apps to use the UDI for all kinds of identification towards services etc. This may lead to hard connection between a user and a device, possibly leakage of who the user really is in a way the user have little, no knowledge nor control of.

This way identification towards a service has to be solved either by deriving an identity based on CDI, or explicitly asking the user for an ID. Which of course could be scripted (having a cookie on the host, for example). But we should at least not provide device specific info to the apps to support this.

quite, 2022-12-08:
Yup, I understand the reasoning.

My notes about the consequences of this change (@mchack-work for information):

This would mean that tkey-ssh-agent only will be able to get hold of the UDI the TKey if is started when TKey is in firmware mode (and it loads the signerapp itself). If signerapp is already loaded when tkey-ssh-agent starts, it would be able to use the agent, but not get hold of UDI for showing through the comment field displayed by `ssh-add -L`.

When tkey-ssh-agent loads the app itself, the UDI is also displayed in the USS input dialog for the user.

Either tkey-ssh-agent won't (or alternatively should not ever) display the UDI in `ssh-add -L` if it doesn't load the signerapp itself. Or maybe it should refuse to operate if it does not find TKey in firmware mode when starting up. Not sure.

secworks, 2022-12-08:
Created a feature branch with a commit that locks down the UDI access:
https://github.com/tillitis/tillitis-key1/compare/main...udi_fw_access_only

quite, 2022-12-08:
Looks good @secworks , will be expecting adjustments to software.md and system_description.md in the same PR ;)

secworks, 2022-12-08:
Good! Yes, those should be updated too. And we need to test the access control too.
I just wanted to add the isse and a commit with the HW changes to have something to discuss.

mchack-work, 2022-12-19:
PLease see PR https://github.com/tillitis/tillitis-key1/pull/60

quite, 2023-01-26:
This was impl in the tk1 core in 5c74a0727c19b4f0f21d75a9eef8e7f231d48a43

closed 2023-01-26 by quite