tillitis/tillitis-key1 #50: Add app access to BLAKE2s in FW

secworks, 2022-12-08:
Add functionality to allow apps to call the BLAKE2s function in the FW for their own use.
The suggested solution is to expose the address to the function entry to the app using a readable 32-bit register. 
The FW will have to get the correct address and write it to the register as part of the FW boot and app load process.

mchack-work, 2022-12-09:
Added support in:
 
- qemu: https://github.com/tillitis/qemu/pull/9
- firmware (and testfw for tests): https://github.com/tillitis/tillitis-key1/pull/53
- rng_stream app: https://github.com/tillitis/tillitis-key1-apps/pull/36

In the apps repo I still have blake2s code left. Please review.

mchack-work, 2022-12-15:
- Removed extra blake2s from -apps.
- Added documentation in main repo.
- Cleaned up a bit after review and squashed. Reviewers agreed that it looks good. Merged.

closed 2022-12-15 by mchack-work

LoupVaillant, 2023-05-29:
Hi,

I noted in the current versions of the docs (and the merge request seems to confirm it), that only the main `blake2s()` function is available. This is in contrast to the source code, which gives access to the full IUF (init, update, final) interface, and the main file even uses.

Perhaps consider reopening this issue to include the IUF interface in addition to the convenience function?

secworks, 2023-05-30:
Currently the function entry point is exposed via a register in the TK1 core. Exposing the IUF would require two more 32-bit registers. This may be possible (we have ~100 registers left). The main issue is not to change either the the FW or the FPGA in a way that would affect the current release. But with the release frozen we could possibly expose IFU through a header file. It would make the life of the new random number generator app easier since it then could just use the Blake2s in the FW. Thanks for making the suggestion!

LoupVaillant, 2023-05-30:
Ah, only 100 registers left, I see why you're being conservative. And keeping ABI compatibility across Tkey versions makes sense. Though I guess that if I can accept tying myself to a given point release, or even an individual TKey I could just find the function I want in the firmware and point to it directly. Not exactly clean, but if it works I'd likely be happy.

I also thought that perhaps you were planning to remove the IUF even internally (makes BLAKE2s a bit simpler, and easier to optimise loading). It would only require a small modification of your main function (just arrange UDS, digest and USS in a contiguous buffer). But that'd forego the ability to hash incoming programs in parallel with the download, assuming the hardware allows it.

secworks, 2023-06-02:
We don't have a pressing need to reduce the FW in size right now.