tillitis/tillitis-key1 #61: Icebram cannot BRAM to replace

0xMihir, 2022-12-28:
When running `make prog_flash`, I get this error during the build process:

```
python3 /Users/mpatil/Documents/Programming/Projects/tillitis-key1/hw/application_fpga/tools/makehex/makehex.py firmware.bin 1536 > firmware.hex
icebram -v bram_fw.hex firmware.hex < application_fpga.asc > application_fpga.asc.tmp
Loaded pattern for 32 bits wide and 1536 words deep memory.
Extracted 192 bit slices from from/to hexfile data.
Found 19 initialized bram cells in asc file.
Found and replaced 0 instances of the memory.
No memory instances were replaced.
```

How do I fix this? I'm building this on a Mac M1.

0xMihir, 2022-12-28:
It seems like this commit was added specifically to cause that error when nothing was replaced. I'm not sure if this is the intended behavior from the TKey build code.
https://github.com/YosysHQ/icestorm/commit/a4d32c985b5432b5a8a3c79b4470a0308a138d1f

0xMihir, 2022-12-28:
Actually, this might be a general build issue on macOS. I tried building the .bin file on a clean linux install and it did replace some portions of memory.

quite, 2023-01-01:
Oh, that's a good change. When icebram fails to replace anything, that is most likely a fatal error. It is also for us.

But can you verify that you're building yosys itself from 06ef3f264afaa3eaeab45cc0404d8006c15f02b1 according to https://github.com/tillitis/tillitis-key1/blob/main/doc/toolchain_setup.md on both Macos and a Linux? We're sitting on that commit to avoid precisely some issue which causes icebram to fail to replace anything.

0xMihir, 2023-01-01:
Sorry! It seems like brew overrode the path for yosys. I deleted the brew one, and it works now.

closed 2023-01-01 by 0xMihir

quite, 2023-01-03:
@0xMihir oh, so you had a yosys installed from brew, which took precedence over the one installed in /usr/local/bin as part of Tillitis toolchain? I guess brew as their own entry in PATH, before /usr/local/bin then? But I don't know or have access to any Macos system. Did you just uninstall yosys from brew? What is the best way to avoid the conflict? Perhaps we should document something...

0xMihir, 2023-01-03:
Yes, I ran `brew uninstall yosys` and reran `sudo make install` after