tillitis/tillitis-key1 #70: Change timer API to have explicit start and stop bits to fix TOCTOU.

secworks, 2023-01-23:
The current API for the timer contain a combined start/stop-bit. Based on the current state (idle or running), writing to the bit will either start or stop the timer. The problem is that when reading the status and then writing to this bit, the status may have changed (the timer finished). A nice example of a TOCTOU issue.

secworks, 2023-01-23:
Commit https://github.com/tillitis/tillitis-key1/commit/b84a5aa18a3ef1d4f116a955247d9e99f3284551 changes the API to have separate 'start' and 'stop' bits.  The start bit is will only work when timer is not running. Similarly, the stop bit will only work when the timer is running.

Additionally, this commit change the name of the 'ready' bit to 'running' to make it more clear. This change also inverse the values. The bit is zero when the timer is not running, and one when the timer is running.

The commit updates the documentation and header file. The commit also updates the core testbench.

quite, 2023-01-23:
@secworks Just to confirm; you mentioned that for the watchdog, writing to the start-bit when the timer is already running would cause it to restart (with same timer value & prescaler). That was only for the watchdog, and not for our general timer then?

secworks, 2023-01-23:
Yes. That is only for the watchdog. Unless you think otherwise. I'm open to suggestions on the API, alining behaviours between cores have benefits (I think - like having control and status on the same intra-addresses.). Would it make sense to have the timer behave like this? When would it make sense? (Use case?)

quite, 2023-01-30:
@secworks i've updated fw & testfw to the new API and pushed a commit. Also small change to software.md in another commit.

secworks, 2023-01-30:
I have reviewed the changes and think this can be merged.

closed 2023-01-30 by quite