tillitis/tillitis-key1 #72: Add watchdog core to tkey FPGA design

secworks, 2023-01-23:
We want to be able to handle TKey FW and apps stuck. We do this by adding a watchdog core.

The core when activated will count cycles from a preset value down to zero. When it reaches zero it will trigger the reset process in the FPGA design. Note that this will NOT force the FPGA device to perform configuration by reading the configuration bitstream, just reset the FPGA design already configured.

FW and SW can avoid the reset by periodically write the 'start'-bit in the watchdog core API. FW and SW can also disable the watchdog by writing to the 'stop'-bit when the watchdog counter is running.

secworks, 2023-01-23:
The https://github.com/tillitis/tillitis-key1/tree/watchdog branch contains a (hopefully) working implementation. It needs review and testing.

mchack-work, 2023-01-26:
When attempting to add support for the hardware watchdog in qemu I noticed that exposing `TK1_MMIO_WATCHDOG_CURR_TIMER` means we have to update it 18 million times per second. This slows down qemu quite a lot.

I propose we don't expose `CURR_TIMER` to software at all.

mchack-work, 2023-01-26:
qemu support in branch watchdog in https://github.com/tillitis/qemu

Still WIP. Timers and MMIO work and the watchdog() callback function is hit but we still can't reset the machine properly.

secworks, 2023-01-27:
Commit https://github.com/tillitis/tillitis-key1/commit/01fab59132375fc7e8c828b0afca35c9160b8e9f remove the CURR_TIMER from the API.

secworks, 2023-02-08:
An important question is if the watchdog improves the ability to perform warm-boot attacks. By crafting an app that simply enables the watchdog after n cycles it allows an attacker to control when a device is being reset, and get the FPGA to configure itself without disconnecting it (and reconnecting it again) to the host.

secworks, 2023-05-03:
Will not be integrated in the TKey1 device.

closed 2023-05-03 by secworks