tillitis/tillitis-key1 #73: Checking if TKey is running an App

0xMihir, 2023-01-23:
Currently, if you try to send app commands when the TKey is in firmware mode, the TKey locks up and cannot load apps. 

It would be nice if there were a way to check that the TKey is currently running an app so you do not need to re-flash software each time.

quite, 2023-01-24:
It would be interesting to know exactly what you send to the firmware which "bricks" it, making app-loading impossible. But I think I can see ways of this happening.

Our usual procedure in the host programs is that they either assume the TKey-app they want to talk to is running, and talk away. This is the simple case.

The `tkey-ssh-agent` tries to probe for firmware by doing GetNameVersion with a read timeout. If it gets a response, then firmware is running and it proceeds to load its signer-app. If it times out, then some app is assumed to be running, so it does GetNameVersion towards the app (which signer-app implements) and if it the signer-app is indeed running, then go ahead and talk to it. If it times out, some other TKey-app is probably running and the TKey currently needs to be removed and plugged in again to put it back in firmware mode.

We're considering to add watchdog which can be used to allow firmware and app to reset itself to a known state if it doesn't receive enough bytes for a cmd etc. https://github.com/tillitis/tillitis-key1/issues/72

quite, 2023-01-24:
Oh, and to specifically answer your question. No you cannot in the general case find out if an app is running, because the app can implement any protocol of choice. So you can try to talk to firmware with timeout, and this *should* fail if an app is running. But it would also fail if you're talking to the wrong serial port/tty. And the app could as well implement the firmware protocol, impersonating the firmware.

Protocol guidelines could be introduced, where TKey app developers agree to implement a certain command in the same way, or to ignore a certain firmware command.

mchack-work, 2023-03-23:
In https://github.com/tillitis/tillitis-key1-apps we decided to probe for firmware and have running apps checking framing protocol header for incoming commands to `DST_FW` and replying `NOK`. Not great, but at least a workaround for now.

Closing.

closed 2023-03-23 by mchack-work