tillitis/tillitis-key1 #76: Add support for ASLR and data protection of the main RAM

secworks, 2023-02-08:
Currently the main RAM has no protection beyond the fact that it is internal to the FPGA. A warm-boot attack, or any other attack that would allow the attacker to read out the contents of the SPRAM blocks would gain access to the contents, including any secrets derived by or handled by an app.

We would like to improve this by adding some sort of address space layout randomization (ASLR). We would also like to scramble the actual data values being stored in RAM. And we want to change the randomization and data scrambling every time the device starts, and also between FW and App modes.

An important limitation is that we can't add any cycles, since this will drastically lower the execution performance. So, whatever solution we come up with must be possible to perform purely combinational, and as part of a normal memory operation. This also without dropping the clock speed to below 18 MHz.

secworks, 2023-02-08:
The branch https://github.com/tillitis/tillitis-key1/tree/ram_scrambling contains a first attempt at adding the address and data scrambling functionality.

We use the Pseudo Random Permutation (PRP) xorshift, with a 32-bit random value as seed. For data we also include the address. This ensures that the same value written to two different addresses will be scrambled differently.

The FW is expected to read out entropy words from the TRNG and write them to the seed registers during boot and before switching from FW mode to App mode.

quite, 2023-03-06:
The second attempt is at https://github.com/tillitis/tillitis-key1/tree/ram_scramble -- i run this on hw and it still works

secworks, 2023-03-07:
There is now a branch with all changes merged into main:
https://github.com/tillitis/tillitis-key1/tree/ram_scramble_rebase

This branch builds, lints. But needs testing and review.

secworks, 2023-03-07:
The branch has been tested with reset values != 0 for aslr and data seeds. And loading and running apps works.

closed 2023-03-07 by secworks