tillitis/tillitis-key1 #79: Programming board: implement faster USB protocol

cibomahto, 2023-02-13:
I'd like to upgrade the USB protocol that the programmer boards speak, in order to have drastically better flash performance. Unfortunately the protocol change is incompatible with the firmware that we currently ship on the programmer boards, so they will need to be updated for this to work.

To upgrade, at the moment you need to:

1. Update the firmware in your programmer to the 'raw_usb' version: https://github.com/blinkinlabs/ice40_flasher/tree/raw_usb (this is already done on the production test jig pico)
2. Update iceprog to the 'raw_usb' version: https://github.com/tillitis/icestorm/tree/raw_usb/iceprog (this should work on Linux, but will need modification again to work on macOs)
3. For anyone using the production test tools (either reset.py, or the actual production test), they will need to switch over to the raw_usb branch in tillits-key1 as well.

Ideally I'd like to merge all of these changes into the main branch on each repo, but doing so will mean that any programmers with older firmware won't work correctly, and will need to be updated.

My proposal is to make this a clean break, and to merge these changes into each of the updated branches. I added a check to iceprog, that warns the user if their programmer firmware needs to be updated. Perhaps this is enough?
[image]

quite, 2023-03-28:
I think this has all been merged?

cibomahto, 2023-03-28:
Yep, seems so!

closed 2023-03-28 by cibomahto