tillitis/tillitis-key1 #81: Move CH552 firmware directory?

cibomahto, 2023-02-14:
The firmware for the CH552 is in a non-obvious location:
/hw/boards/mta1-usb-v1/ch552_fw

Is there a better place that this could go? It's also used in the tkey design ( /how/boards/tk1 ), so I don't think the current location makes sense any more. I put it there originally to copy how the application fpga firmware is stored ( /hw/application_fpga/fw ).

secworks, 2023-02-15:
I agree. I think It should be located in a subdir to the board it is being used on.

cibomahto, 2023-02-15:
It was used on MTA1, and now also in TK1, and possibly in a future board as well?

We could create copies of it, and that would be a way to freeze the version to what shipped on the board.

closed 2024-03-27 by dehanj