tillitis/tillitis-key1 #85: Byte-adressable ROM

quite, 2023-02-24:
We'd like to hash/sign our firmware residing in ROM. The functions for doing that requires byte-adressable memory, which ROM currently is not.

secworks, 2023-02-24:
I will look at this. It should be quite easy to fix.

quite, 2023-02-24:
Actually the work-around of first copying to RAM is not too bad... but surely it would look neater to work directly on ROM

secworks, 2023-02-27:
Updates the name to reflect that this seems not only to be about the ROM, but also FW-RAM, and potentially other mem areas.

secworks, 2023-02-27:
Branch created: https://github.com/tillitis/tillitis-key1/tree/byte_access

quite, 2023-02-28:
Byte-reads from ROM works. Verified by actually pointing crypto_sha512() at addr 0x00. It produces known expected hash.

And actually byte-reads from TK1_MMIO_TK1_CDI_FIRST gives the same result as 32-bit word-reads. Tested in our testfw.

closed 2023-02-28 by quite