tillitis/tillitis-key1 #87: Handling of illegal instructions

secworks, 2023-02-27:


secworks, 2023-02-27:
Digging through the code for the PicoRV32: https://github.com/tillitis/tillitis-key1/blob/main/hw/application_fpga/core/picorv32/rtl/picorv32.v

One can see that if the CATCH_ILLINSN option is set (it is set by default), but no IRQs are enabled (they are not enabled by default), the CPU will move to the cpu_state_trap (see line 1491). In this state, the CPU will set the trap signal (which is an external port), and then stay in the state until reset. This means that in our instance of the CPU, any illegal instructions will hang the CPU at the instruction that was decoded. Not really clear what happens if CATCH_ILLSN is not set. The CPU probably executes something. Which we don't want.

What we could do is have hardware that observes the trap signal, and if being set starts flashing the LEDs.

secworks, 2023-02-27:
The following commit contains functionality to look at the trap signal being set. And if it is set will flash the LEDs until the device is reset.

https://github.com/tillitis/tillitis-key1/compare/main...trap_indicator

quite, 2023-03-06:
I've tried executing 0x0000cafe and got a nice red flash!

secworks, 2023-03-06:
I'd say we merge this into main.

quite, 2023-03-06:
@secworks agreed

secworks, 2023-03-06:
Merged.

closed 2023-03-06 by secworks