tillitis/tillitis-key1 #98: Document NVCM programming procedure

cibomahto, 2023-03-08:
First, the script needs to be run in a python virtual environment. It can use the same one as the production test, so this would be sufficient to set it up:

    ./run (then ctrl+c to exit the production test once it's set up)
    . venv/bin/activate

Next, there are three separate steps that need to be performed for NVCM programming: write, verify, then secure. Note that the device will boot from nvcm once the write stage is completed, but it will still allow read backs until the secure command is completed.

    ./pynvcm.py --write ../application_fpga/application_fpga.bin --my-design-is-good-enough
    ./pynvcm.py --verify ../application_fpga/application_fpga.bin
    ./pynvcm.py --secure --my-design-is-good-enough

dehanj, 2023-08-29:
@cibomahto i'm about to break out the pynvcm.py with friends to a separate repository to separate it a bit from the rest in this repo. 
Just to cross-check, it is still fine to do the write, verify and secure in one go? It seems to work fine during my tests, but got a bit curious since you mention it as three seperate steps here
```
./pynvcm.py --write ../application_fpga/application_fpga.bin --verify ../application_fpga/application_fpga.bin --secure --my-design-is-good-enough
```

cibomahto, 2023-08-30:
I think it's fine, the Lattice tools don't power cycle the chip between these steps.

Just note that pynvcm doesn't honor the order of operations passed to it; it has a fixed order that it evaluates them (which works for this case):
https://github.com/tillitis/tillitis-key1/blob/7cd085a17e2b7849ad4ff192286a65678412d2eb/hw/production_test/pynvcm.py#L725

I think I wrote it as separate steps for clarity.

dehanj, 2023-08-31:
Thanks, appreciate the clarity!

dehanj, 2023-09-11:
pynvcm.py now lives in its own repo: https://github.com/tillitis/pynvcm

We have not removed these files from this repo, but it will most likely be done in the near future, so closing this.

closed 2023-09-11 by dehanj