tillitis/tillitis-key1 #110: cannot program again after aborting with ctrl-c

quite, 2023-03-27:
If I run tillitis-iceprog, and abort it using ctrl-c, then i cannot run it again -- it hangs after saying it has raw power. It does not matter if there is a usb stick sitting in the jig or not. The programmer has to be re-plugged to get it working again.

cibomahto, 2023-03-27:
This is probably because the usb device isn't released properly when the program is interrupted by SIGINT.

I think the correct way to fix that is (in rpi_pico_interface.c) to install a signal handler after iceprog claims the device, something like this:

```
void close_on_sigint(int unused) {
    close(); // Call the close routine to release the USB device
}
```

And register the handler before [printing the 'raw power' message](https://github.com/tillitis/icestorm/blob/1d3351826dfbb58133a0a44be91814a78c5f2ff8/iceprog/rpi_pico_interface.c#L364) :

   signal(SIGINT, close_on_sigint);

The previous communication protocol used HID, and this might have been handled automatically by the hidapi library, or might not have been needed if it used the kernel HID interface instead of the USB interface.

quite, 2023-03-27:
doing the `close()` on `SIGINT` also does not help.

quite, 2023-03-27:
Can you reproduce? Maybe this is on my AMD laptop, that also doesn't benefit from any speed increase heh

quite, 2023-03-28:
Seems ctrl-c:ing does not cause problems on Intel laptop

quite, 2023-03-28:
seems a call to `flash_release_reset()` was what was needed. But I think i'm gonna catch SIGINT in main, let the handler set a bool and let while/for loops stop when that bool is true. And if interrupted probably exit the program with code 1.

closed 2023-03-30 by quite