tillitis/tillitis-key1 #127: Debug Verilator simulation

secworks, 2023-04-27:
The Verilator top level simulation seems not to work. We should debug and then add it to the CI flow.

secworks, 2023-06-29:
The verilator simulation builds and starts. It can respond to signals (as touch events). It also opens a pty that should be possible to use to load apps. @mchack-work  will try and do that. If that works, it seems the Verilator model works.

The next step would be to decide how to integrate into the CI-flow. For example load a simple app.

secworks, 2023-09-11:
Reassign to MC.

mchack-work, 2023-09-13:
The Verilator (version  5.014) build stops with these complaints:

```
%Warning-UNOPTFLAT: /home/mc/tillitis/git/tillitis-key1/hw/application_fpga/rtl/rom.v:46:16: Signal unoptimizable: Circular combinational logic: 'application_fpga.rom_inst.rom_ready'
   46 |   reg          rom_ready;
      |                ^~~~~~~~~
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=5.014
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    /home/mc/tillitis/git/tillitis-key1/hw/application_fpga/rtl/rom.v:46:16:      Example path: application_fpga.rom_inst.rom_ready
                    /home/mc/tillitis/git/tillitis-key1/hw/application_fpga/tb/application_fpga_vsim.v:345:3:      Example path: ALWAYS
                    /home/mc/tillitis/git/tillitis-key1/hw/application_fpga/tb/application_fpga_vsim.v:92:17:      Example path: application_fpga.rom_cs
                    /home/mc/tillitis/git/tillitis-key1/hw/application_fpga/rtl/rom.v:59:3:      Example path: ALWAYS
                    /home/mc/tillitis/git/tillitis-key1/hw/application_fpga/rtl/rom.v:46:16:      Example path: application_fpga.rom_inst.rom_ready
```

but building with `-Wno-fatal` works. However running the resulting simulation yields disappointing results:

```
% tkey-runapp --port /dev/pts/5 apps/blink/app.bin 
Connecting to device on serial port /dev/pts/5 ...
GetNameVersion failed: ReadFrame: Read timeout
```

dehanj, 2024-06-17:
Did a quick check, the warnings does not seem to be present anymore while building. 
I have not tried to communicate with the port it gives.

But is this something that we want to finish or should we close and deprecate this?

mchack-work, 2024-06-18:
Having a full simulator in software seems very useful for me. I don't think it's a top priority, but it would certainly help.

I can also build without warnings now. However...

```
% ./verilated/Vapplication_fpga 
                                                                            
generate touch event: "$ kill -USR1 202223"
pty: /dev/pts/9

% tkey-sign --port /dev/pts/9 -G -p key.pub 
WARNING: App already loaded.
GetAppNameVersion: ReadFrame: Read timeout
Couldn't load signer: no TKey on the serial port, or it's running wrong app (and is not in firmware mode)
```

dehanj, 2024-11-12:
This might not be the only problem, but at least the uart is configured wrong for the Verilator simluation.

In `hw/application_fpga/tb/application_fpga_verilator.cc`
```
// Clock: 18 MHz, 62500 bps
// Divisor = 18E6 / 62500 = 288
#define BIT_DIV 288
```

dehanj, 2024-12-09:
Seem to be fixed with the PR linked above, albeit it takes an awful amount of time to load an app (like 4 minutes).
A solution for that is worked on separately.

closed 2024-12-09 by dehanj