tillitis/tillitis-key1 #141: USB-A version ?

udf2457, 2023-08-21:
Thank you for all your work on TKey.

Even if you do not plan to release it in off-the-shelf form, it would be appreciated for those of us who are not PCB designers if you could at some point update the resources to add a modified design with USB-A.

Reason behind the request is that it can be envisaged that being able to plug-in a Tkey to servers could be a useful use-case.  Most (all ?) servers do not come with USB-C, and the prospect of USB-C to USB-A dongles is unattractive for obvious reasons.

dehanj, 2023-08-24:
We definitely have this use case in mind, but we don't have any firm plans as of now. 
Appreciate the feedback, glad you like TKey.

udf2457, 2023-08-24:
Thank @dehanj !   

P.S. If you do ever do it, please remember to try to keep the final design slim enough so it can fit on a server PCB slot.  :wink:

closed 2023-08-30 by secworks