tillitis/tillitis-key1 #144: possible compatible hardware: FOMU

coelner, 2023-09-01:
Maybe this is also a capable device to use:

https://tomu.im/fomu.html
https://www.crowdsupply.com/sutajio-kosagi/fomu#products
https://workshop.fomu.im/en/latest/background.html

    FPGA: Lattice ICE40UP5K
    Speed: 48 MHz external oscillator
    RAM: 128 kB RAM for a soft CPU¹
    Storage: 2 MB SPI flash
    Connectivity: USB 2.0 FS (12 Mbps)
    Buttons: Four
    LEDs: One RGB

rel: https://github.com/tillitis/tillitis-key1/issues/141

The CH552 is missing, but AFAIK it is integrated into the FPGA via tinyUSB (https://github.com/hathach/tinyusb)

secworks, 2023-09-04:
Hi!

Yes, the FOMU is close to the TKey1 in terms of functionality.
Unfortunately, you will not be able to fit the TKey1 FPGA design and the TinyUSB core in the same device. We looked at using TinyUSB or another open FPGA based USB solution on the TKey1. (One can still see this in the name "application_fpga" for the FPGA design in TKey1, there would have been an "interface_fpga" in front of it. Due to reliabilty issues and time constraints, we had to change to the CH552.)

coelner, 2023-09-04:
Well, too bad. But thanks for the explanation.

closed 2023-09-04 by coelner