tillitis/tillitis-key1 #153: System architecture questions

MuratovAS, 2024-01-14:
Good afternoon.
What motivated the decision to implement your own protocol for communicating with the host? You could have used `ccid`. This would improve compatibility.

I understand that the concept of generating a key based on `uds` is different from simply storing the key (for example, as in `gnuk`). Even the read-only implementation of `OpenPGP Smartcard` allowed the device to be used in more scenarios.

To work, you need to override `ssh agent`. What blocks the use of the rest of my devices gnuk, yunikey

Despite all of the above. The device seems interesting as `u2f`.

mchack-work, 2024-01-16:
The current USB chip doesn't support CCID.

Why did we choose that chip? I think it was some of us had experience with it and thought it would be easy to work with. We were rather rushed to get something out the door to see if it would work.

The goal wasn't necessarily to create a security token or something like a smart card. The goal was, at least initially, to create a small secure computing evironment with unconditional measured boot.

Since then a lot has happened and we have been made aware of the need for a secure computing environment like ours that would be more compatible with tokens and smart cards available over USB HID and CCID, but the current generation doesn't support that, except for the proof of concept U2F thing that only works on Linux and fakes a HID on the client side, just to show it can be done:

https://github.com/tillitis/tkey-fido

If you want to use different ssh agents, tkey-ssh-agent, gpg-agent, and OpenSSH's standard ssh-agent at the same time it's quite possible. Either you set an environment variable pointing to the Unix socket of the agent you would like to use or you add it to your `.ssh/config` like this:

```
Host github.com
     IdentityAgent ${XDG_RUNTIME_DIR}/tkey-ssh-agent/sock
```

if you want to use tkey-ssh-agent for all access to github.com, for instance.

MuratovAS, 2024-01-16:
Thank you for what you do.
Thanks for the detailed answer.

closed 2024-01-16 by MuratovAS