tillitis/tillitis-key1 #162: fw: Add randomness to memory

mchack-work, 2024-03-05:
The firmware at start first does a complete wipe of the memory in the `crt0.S` in assembly, then in `scramble_ram()` it turns on the RAM address and memory scrambling so an outside observer has to do more work to get at the content. Then it also fills the entire memory with a random word from the TRNG. The same random word for the entire RAM.

It would probably be better to randomize the entire RAM content. However, the TRNG is slow, so we need some other way of doing this. Perhaps by using our `blake2s()` and hashing over something like something from the TRNG and something, like the address?

dehanj, 2024-03-21:
> The same random word for the entire RAM.

This is not quite correct.

```
    ...
        uint32_t rnd = rnd_word();
        uint32_t rnd_incr = rnd_word();
    ... 
        for (uint32_t w = 0; w < TK1_RAM_SIZE / 4; w++) {
            ram[w] = rnd;
            rnd += rnd_incr;
        }
```

The ram is filled with random words, that is incremented with a random word for every loop. The same random word in every loop. This is, however, still quite easy to break. Simply compare two ram addresses and you have the increment used. Takes some more effort, but not much. I would argue the `blake2s()` approach would considerably make it harder to guess the content.

secworks, 2024-04-15:
PR https://github.com/tillitis/tillitis-key1/pull/201 adds randomness.

PR #201 (2024-06-11): xorwow based initial RAM randomization
This PR improve initial RAM randomization by adding the xorwow variant of Marasaglia xorshift PRNG as data generator.

Closes #162
closed 2024-06-11 by dehanj