tillitis/tillitis-key1 #176: Add security linters to CI

secworks, 2024-03-19:
There are a few interesting linters, STAs that we could add to the CI flow to improve checks:

Flawfinder
https://dwheeler.com/flawfinder/

splint
https://splint.org/

scan-build - exists as action in github
https://clang-analyzer.llvm.org/scan-build.html

mchack-work, 2024-03-21:
I agree more analysis tools would be good.

splint is already added to check at least firmware code. See the "check"
target in `hw/application_fpga/Makefile` which also calls clang-tidy
with at least the `cert-*` checks.

splint currently gives us a lot of warnings, mostly because we haven't
added all the formal comments that splint wants us to insert in our
code, but there may be nuggets here. We pass clang-tidy's check.

However, splint, as you know, seems dead as a project and clang-tidy
gives us much more.

The `make check` isn't in CI, so yes, very good idea to add it, but
perhaps either delete splint or carefully go through its output.

SonarQube's cloud offering seems to be available free for FLOSS
projects. We probably qualify and should definitely look into it.

scan-build is new to me. I have to look into it.

dehanj, 2024-10-10:
Here is a suggestion from our friends across the river: https://github.com/Ericsson/codechecker

dehanj, 2026-09-24:
Duplicate of #332. Closing.

closed 2026-09-24 by dehanj