tillitis/tillitis-key1 #196: Increase clock frequency of application_fpga

secworks, 2024-03-26:
The application_fpga currently meets 24 MHz after P&R, but is clocked at 18 MHz. We should be able to increase the clock to ~22 MHz and still have good timing margin. This would improve performance with 20+ percent.

What we need to do:
- Update the PLL parameters
- Update build scripts
- Update UART baud rate calculationa
- Update any other timing - timemouts, LED blinkenlights timing etc.

secworks, 2024-04-15:
The branch https://github.com/tillitis/tillitis-key1/tree/Increased_clock_frequency contains a change to the PLL settings to run clock at 21 MHz. It builds an meet timing. But we should add all features we want, need and then try to increase the clock speed.

dehanj, 2024-06-17:
These are the baudrates that the CH552 can work with:
These are the true baudrate, so 0% errors. I believe up to 5% error i usually OK.
Divisior : buadrate
8 : 125 000
6 : 166 666
5 : 200 000
4 : 250 000
3 : 333 333
2 : 500 k Buad
1 : 1 M buad (not confirmed to work on CH552, so I would say it is out of scope)

closed 2024-08-20 by dehanj