tillitis/tillitis-key1 #219: Change names of ADDR_RAM_ASLR and ADDR_RAM_SCRAMBLE

mchack-work, 2024-04-25:
To remove potentially confusing users we might want to change some names.

Quoting @secworks

> I suggest that we change both address and data randomization to ADDR_RAM_ADDR_RAND and ADDR_RAM_DATA_RAND. We are performing basically the same randomization operation. for both. Scrambling is possibly more correct now, but is longer. And if move to a PRNG it is randomization.

This changes both Verilog and `tk1_mem.h`. In the header file we have:

```
// Deprecated - use _ADDR_RAND instead
#define TK1_MMIO_TK1_RAM_ASLR 0xff000100

#define TK1_MMIO_TK1_RAM_ADDR_RAND 0xff000100
#define TK1_MMIO_TK1_RAM_SCRAMBLE 0xff000104
```

We will probably keep the old names for a while, but add the new names, too, just like the old `RAM_ASLR`.

secworks, 2024-04-25:
What I suggest is:

```
// Deprecated - use _ADDR_RAND and _DATA_RAND instead
#define TK1_MMIO_TK1_RAM_ASLR 0xff000100
#define TK1_MMIO_TK1_RAM_SCRAMBLE 0xff000104
 
#define TK1_MMIO_TK1_RAM_ADDR_RAND 0xff000100
#define TK1_MMIO_TK1_RAM_DATA_RAND 0xff000104
```

secworks, 2024-04-25:
BTW: Why do we have TK1 twice in the names?

mchack-work, 2024-04-25:
I believe the first `TK1` is a prefix to tell the developer these are constants that work with the TK1 hardware. The next `TK1` is the core. Compare:

```
#define TK1_MMIO_TRNG_STATUS 0xc0000024
```

See 

https://github.com/tillitis/tillitis-key1/blob/main/hw/application_fpga/fw/tk1_mem.h

secworks, 2024-04-25:
Yeah, realized it directly and removed the comment.

> I believe the first `TK1` is a prefix to tell the developer these are constants that work with the TK1 hardware. The next `TK1` is the core. Compare:
> 
> ```
> #define TK1_MMIO_TRNG_STATUS 0xc0000024
> ```
> 
> where `TRNG` is the core.
> 
> See
> 
> https://github.com/tillitis/tillitis-key1/blob/main/hw/application_fpga/fw/tk1_mem.h
> 
> for the whole shebang.

closed 2024-08-20 by dehanj