tillitis/tillitis-key1 #272: Splitting tkey-builder into multiple OCI images

mchack-work, 2024-10-14:
The tkey-builder OCI image has all development tools for FPGA development, for software development of firmware and device apps, and development of the TP1 firmware. This makes it unnecessary big (over 5 GiB). The most common use case is probably client/device app development and we should streamline for this use.

We propose to split tkey-builder into several images with specific intended use:

- [ ] tkey-app-builder

 In the tkey-devtools repo for development of:

 - client apps (go etc)
 - device apps (clang etc)

 Possibly based on https://github.com/silkeh/docker-clang

- [ ] tkey-builder

 Stays in the tillitis-key1 repo for:

 - Bitstream tools (yosys/nxtpnr etc)
 - firmware development (clang etc)

- [ ] tkey-programmer-builder

 In the tp1 repo:

 - Development of tp1 firmware (arm-gcc)
 - Pico library (pico-sdk)

mchack-work, 2024-10-22:
I pushed a first version of the tkey-app-builder to tillitis-key1 but maybe it's better to put it in tkey-devtools like the qemu image?

dehanj, 2024-10-23:
I agree that it is better that it lives in tkey-devtools. Makes more sense then tillitis-key1.