tillitis/tillitis-key1 #275: fw: Persistent storage syscalls

mchack-work, 2024-10-14:
Placeholder!

Implement persistent storage syscalls:

1. Retrieve information about if the device app has an allocated
   storage area or not.

2. Allocate or deallocate a storage area. The FW shall return status
   indicating if the allocation was successful or not.

3. Perform read or write operations of data to the allocated storage
   area.

dehanj, 2024-10-31:
Unimplemented features and/or open questions

- USS for a preloaded app.
The preloaded app should be able to use the USS.
The USS then needs to be stored in the partition table, in order for
the pre-loaded app to be able to start up without a user supplying the
USS on every boot. It could also mean an option to enable that a user
has to supply the USS at each boot. This is still an open question.
How do we securely store the USS in flash?

- Auto start of a pre-loaded device app
1) auto-start (unless a cmd comes), 2) Wait for cmd, either start pre-loaded or load another app.
Can be configurable when installing a pre-loaded app.

- App digest from device to client when installing a pre-loaded app

- Management app privileges
How much power should the management app have? In the current PoC it can only register/unregister itself and store/delete a preloaded app. Should it be able to
1) erase arbitrarily storage area?
2) manipulate the partition table?
3) "Factory reset"?
4) Access the entire storage?

Some of these might be useful if someone wants to completely erase the storage or have forgotten the USS to an app that has an allocated area.

- Redundancy of the partition table in flash
Write to alternative pages, if something goes wrong during a write it is possible to restore to the previous and only the "latest" change is lost.

mchack-work, 2025-02-28:
Updated the main issue text since we're now resuming work on this.

The new syscall mechanism #305 has been merged and we will use code from #298 on a new branch to adopt the storage to a new world.

mchack-work, 2025-02-28:
Suggested order of commits:

- Add SPI and flash code.
- Add filesystem parts (partition table)
- Copy code to start app from flash - change state machine to include states to check for resetinfo and loading from flash.
- Add storage syscalls for device app.

mchack-work, 2025-04-04:
Implemented:
 
- Reset intention handling.
- Start device apps from flash. Two app slots supported.
- Partition table checksum and redundancy.
- Loader app for testing implemented. Chaining to verified blink app works, so a combination of measured boot + verified boot works.
- 4 device app storage areas with 128 k each.
- Syscalls to support all the above.

closed 2025-05-06 by mchack-work