tillitis/tillitis-key1 #277: Dockerfile: Make more resilient

mchack-work, 2024-10-17:
Not very happy with code patching, even if it's small, in the Dockerfile but since things are moving so fast I don't see any alternatives either than maintaing a fork of nextpnr of our own. That seems like too much work. We will have to revisit this later, I think.

I think some of the git clones can be replaced by using actual release tar balls sometimes, which might be preferable. We might also want to include digests to see that the downloaded file is what we expected.

dehanj, 2024-10-17:
> Not very happy with the code patching of nextpnr, even if it's small, in the Dockerfile. Alternatives? Keep a fork of our own of nextpnr? Anything else?

I guess the options we have is, in some form of sorted list where 1 is the "best" option:
1. upstream the changes
2. figure out if it is really needed, is there any workarounds we can use outside of the specific software?
3. If the change is small enough: patch in dockerfile
4. fork the project and maintain

mchack-work, 2025-02-11:
Added the `contrib/buildtools.sh` script that explicitly checks digests or specific commit hashes of everything. I think that's at least a start of knowing exactly what we include.

I think that makes me at least happier. Some local changes to nextpnr, still, but let's keep tracking development there.

Closing.

closed 2025-02-11 by mchack-work