tillitis/tillitis-key1 #331: Standardise reset request in protocol

mchack-work, 2025-03-14:
In Castor device apps are in control of the reset function. They can reset whenever they want. If a *client* wants the TKey to reset they have to request this from the device app.

In order to create interoperability between any combination of client app and any device app that is running I propose we standardise a way of doing a reset protocol request.

The firmware might also support this protocol request and we may want to at least discuss what should happen when it receives such a request.

What kind of reset intentions do we allow a client to do?

We support these:

1. Start from flash slot 1 (default).
2. Load and start an app from flash slot 1 with a specific app hash.
3. Load and start an app from flash slot 2 with a specific app hash.
4. Load and start a new app from client.
5. load and start an app from client with a specific app hash.

mchack-work, 2025-03-14:
Some possible solutions:

- Make it a special treatment of the framing protocol, just like the firmware probe? The firmware probe is a framing header with endpoint 2 (firmware) and all well-behaved apps answer back NOK.
- Add something to the reserved bits in the framing protocol?
- Use something in the app level protocols? Framing header with endpoint either 2 (fw) or 3 (dev app) + RESET_CODE and reserve the reset codes not to be used for anything else.

mchack-work, 2025-09-08:
During a reset flow discussion today it was decided that we want to support client reset -> start loader with verified boot on app slot 0 -> device app from client.

This means we need to support some additional data in a client reset request, at least:

1. Start from app slot 0.
2. Digest of device app from client.
3. Signature over this digest.

All this data must be left by the running device app for the loader to use.

dehanj, 2026-09-15:
It is decided that the standardized reset protocol is 

| *command*             | *function*              | *length* | *code* | *data*                              | *response*            |
|-----------------------|-------------------------|----------|--------|-------------------------------------|-----------------------|
| `CMD_RESET`           | Reset TKey              | 128 B      | 0xfe   | 1 B reset type, 126 B next app data | none                  |

With the reset types defined in firmware.

closed 2026-09-16 by dehanj