tillitis/tillitis-key1 #357: Flashed Castor on my TKey and stuck with bit #7 non zero - have I bricked it?

ludblom, 2025-06-20:
Tried to flash https://github.com/tillitis/tillitis-key1/tree/TK1-Castor-alpha-1 on my unlocked TKey. I followed the instructions in the README and ran

`cd contrib
make flash`

Assumed it was the FPGA option that I should use and not the USB controller option but guess I mistook that.

The output I now get is the same as when I try to load a new app without unplugging the device, ie

> Auto-detected serial port /dev/ttyACM0
> Connecting to device on serial port /dev/ttyACM0 ...
> GetNameVersion failed: ReadFrame: Couldn't parse framing header: reserved bit #7 is not zero
> If the serial port is correct, then the TKey might not be in firmware-
> mode, and have an app running already. Please unplug and plug it in again.

So my question is, have I bricked my TKey or can I save it somehow?

ludblom, 2025-06-24:
But this is sort of the issue and hence why I believe it may be bricked, for me to flash it with a version I know is working, then the key have to be in flashing mode, but it is permanently stuck in application mode even though I have not loaded an application, so hence I can not flash it with a new version.

agren, 2025-06-25:
Glad you got it working!